CelliesProjects/aquacontrol32 #4: #define SKETCH_VERSION                     "0.0.2" gets stuck

Odaller, 2018-01-12:
Compiling and uploading sketch "version 0.0.2" runs flawlessly.
Start up of controller seems to get stuck. Serial monitor output:

ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11392
entry 0x40078a9c

CelliesProjects, 2018-01-13:
Serial output is replaced by `ESP_LOGx` calls.

All development is done for the `ESP32 Dev Module` which has support for `ESP_LOGx` macros.

I don't want serial statements scattered through all of my code, as it takes up ram and cpu resources.
The `ESP_LOGx'  statements are only compiled (added to the code)  when you select a appropriate log level.
See the [ESP IDF docs](http://esp-idf.readthedocs.io/en/latest/api-reference/system/log.html) for more info.

Which means you have to compile aquacontrol for the `ESP32 Dev Module` and select your desired log level in the Arduino IDE `Tools` menu.

While setting up your hardware etc, you could use level `Info` or `Debug` to show what happens.
The executable code will be bigger, slower and less deterministic, but you can see what is going on.

Once you are happy with your setup, you can compile for debug level `None` and your executable will be smaller and faster.

Odaller, 2018-01-14:
I set the board already to ESP32 Dev Module before your answer.
Today, I downloaded the new code and I'm happy to say it works!

closed 2018-01-15 by Odaller