CelliesProjects/aquacontrol32 #7: Feature check

Allhappy, 2018-04-27:
Hello, thank you very much for your project, I´m testing it, and for now it is great!

I would like to know the maximum number of channels, and if is possible to add channels, for simple ON/OFF (with timers)? 
I have tested and I can switch it without "fadein/fadeout" but I dont have a button on web for single (temporary) on/off.

My current setup is:
 - lights with 3 channels controlled by pwm
 - 3 temperature sensors DS18B20 (Water, Leds, pwm drivers)
 - 8 relay board for 230V switch, 4 currently in use (Heater, Filter, CO2, Skimmer)

I´m electronic technician , and my programming skills are limited for now, a help is much appreciated ! :)

Thanks in advance!

CelliesProjects, 2018-04-27:
Many thanks you for your appreciation! 

I am not planning on adding any channels and on/off switches, sorry. 
Maybe I might add a temperature controlled output to switch a heater, but there are other plans that I want to finish first.
And other projects and real life want some time too. 8D
Second question is a bit unclear, not sure I understand that request. (about the fadein/fadeout)

Sounds like a serious setup you have. I take it it is a saltwater tank?

Allhappy, 2018-04-30:
Hi, 
I have a planted low tech aquarium, I will take a photos to show my project, 

Is a modest project, based on a arduino mega, display, rotary encoder and a "menu code" that I adjusted the needs and it was running non stop for 2 years!
 But I want more :) the esp32 have the wireless  ability, and your project is 80% what I need !!!
I will try to make a few adjustments.

When I said "fadein/fadeout", I mean "sunrise/sunset" :P I mean, rather than using a channel to a led, using it to a Relay... from 0% to 100%. but, this is not a full resolution, because I need two more channels.  
But this feature is very interesting  (temperature controlled output to switch a heater) and at this time it will be 99% what I need :) 
  Unhurried, fortunately all of us have real life, It is a good sign!

I'm available for testing your project. I'm sorry for my bad English :)

Best regards from Portugal.

CelliesProjects, 2019-02-16:
Hi Allhappy.
I just discovered this repo, it might be of interest to you.
https://github.com/mathompl/AquaLed
LEDs, relays and sensors.

closed 2019-02-16 by CelliesProjects

Allhappy, 2019-02-18:
Hi CelliesProjects,
Thanks for the link, I will take a look at it ;)
unfortunately, there is no wireless ... the ESP32 is a best choice.

Thanks!