CelliesProjects/aquacontrol32 #9: use of enum 'lightStatus_t' without previous declaration

tatasker, 2018-09-24:
I keep running into this error and searching the web for a solution but to know avail.

I am using Arduino IDE 1.8.7 (Windows Store 1.18.15) and the latest version of Aquacontrol32.

Below is the error message log.

Arduino: 1.8.7 (Windows Store 1.8.15.0) (Windows 10), Board: "LOLIN D32, Default, 80MHz, 921600, None"

aquacontrol32:114:6: error: use of enum 'lightStatus_t' without previous declaration

 enum lightStatus_t

      ^

aquacontrol32:119:39: error: 'lightStatus_t' does not name a type

 const char lightStatusToString( const lightStatus_t status )

                                       ^

aquacontrol32:119:60: error: two or more data types in declaration of 'lightStatusToString'

 const char lightStatusToString( const lightStatus_t status )

                                                            ^

aquacontrol32:115:1: error: expected unqualified-id before '{' token

 {

 ^

aquacontrol32:119:39: error: 'lightStatus_t' does not name a type

 const char lightStatusToString( const lightStatus_t status )

                                       ^

C:\Users\Tom\Documents\Arduino\aquacontrol32\aquacontrol32.ino: In function 'const char lightStatusToString(int)':

aquacontrol32:123:10: error: 'LIGHTS_OFF' was not declared in this scope

     case LIGHTS_OFF:   return PSTR(" LIGHTS OFF");

          ^

In file included from C:\Users\Tom\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0\cores\esp32/WString.h:29:0,

                 from C:\Users\Tom\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0\cores\esp32/Arduino.h:150,

                 from sketch\aquacontrol32.ino.cpp:1:

C:\Users\Tom\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0\cores\esp32/pgmspace.h:36:25: error: invalid conversion from 'const char*' to 'char' [-fpermissive]

 #define PSTR(s)       (s)

                         ^

C:\Users\Tom\Documents\Arduino\aquacontrol32\aquacontrol32.ino:123:31: note: in expansion of macro 'PSTR'

     case LIGHTS_OFF:   return PSTR(" LIGHTS OFF");

                               ^

aquacontrol32:124:10: error: 'LIGHTS_ON' was not declared in this scope

     case LIGHTS_ON:    return PSTR(" LIGHTS ON ");

          ^

[... 238 lines cut ...]
       return request->send( 200, HTML_HEADER, lightStatusToString( lightStatus ) );

                                                                    ^

webservertask:474:82: error: return-statement with a value, in function returning 'void' [-fpermissive]

       return request->send( 200, HTML_HEADER, lightStatusToString( lightStatus ) );

                                                                                  ^

Multiple libraries were found for "SSD1306.h"
 Used: C:\Users\Tom\Documents\Arduino\libraries\ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display
 Not used: C:\Users\Tom\Documents\Arduino\libraries\esp8266-oled-ssd1306-3.2.5
 Not used: C:\Users\Tom\Documents\Arduino\libraries\arduino_880236
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\Tom\Documents\ArduinoData\packages\esp32\hardware\esp32\1.0.0\libraries\WiFi
 Not used: C:\Program Files\WindowsApps\ArduinoLLC.ArduinoIDE_1.8.15.0_x86__mdqgnx93n4wtt\libraries\WiFi
exit status 1
use of enum 'lightStatus_t' without previous declaration

This report would have more information with
"Show verbose output during compilation"
option enabled in File -> Preferences.

Please advise.

CelliesProjects, 2018-09-26:
I will look into this.

CelliesProjects, 2018-09-28:
This issue should be fixed with 12f425399c4a6f3f1bfe6d100e0aef7247627ed6.
Still no W10 pc to test this, so some feedback would be welcome.

tatasker, 2018-09-28:
I will get to it later today.

CelliesProjects, 2018-09-29:
Just tested it on a borrowed W10 laptop and https://github.com/CelliesProjects/aquacontrol32/commit/12f425399c4a6f3f1bfe6d100e0aef7247627ed6 seems to work fine.

CelliesProjects, 2018-10-11:
The error 
```
C:\Users\Tom\Documents\Arduino\aquacontrol32\aquacontrol32.ino: In function 'void setup()':

aquacontrol32:234:33: error: no matching function for call to 'Adafruit_ILI9341::begin(int, SPIClass&)'

tft.begin( TFT_SPI_CLOCK, SPI );
```
is because the ILI9341 library you are using is not the correct version.
Check [ your included libs](https://github.com/CelliesProjects/aquacontrol32/blob/master/libraries.md ).

tatasker, 2018-10-11:
Thank you I am going to test it as soon as I can get to it.

closed 2019-04-20 by CelliesProjects