CelliesProjects/aquacontrol32 #11: Oled random pixels when powered by 5v directly

Odaller, 2019-04-14:
When I power the ESP32 with the usb cable and a wall socket, the oled screen boots up as it should and the text appears smoothly.

When I remove the usb cable and put 5v directly on the VCC input, the ESP boots up normally, and I can get to the webpage, but the OLED screen only shows random pixels.

I think something is wrong with the initialization of the screen, but I cannot figure out what it is.

CelliesProjects, 2019-04-20:
Probably not the cause, but you are feeding 3v3 to the oled right?
Any changes in the software you suspect could have caused this?

CelliesProjects, 2019-04-21:
I just solved a potential problem that might be related.
The TFT screen was refreshing at 25Hz, which is way too much. 
Just try and pull 7ce82acce981862bee0d9def62f8aa590baa8960 to see if it solves your problem.

CelliesProjects, 2019-04-21:
Oops wrong button. 8P

closed 2019-05-05 by CelliesProjects