CelliesProjects/aquacontrol32 #32: Cooling Fan integration

lullaby37, 2019-08-13:
Hey,
nice project! I have built the aquacontrol2 on the ESP8266 and I am now aiming for the aquacontrol32, I am just waiting for my ESP32 to arrive.
Anyways, your project is awesome and I wish I could add stuff but my programming skills are just too basic and this is much too complex and advance for me, so this not really an issue but rather a request or a wish for future versions. I hope this is okay.
As you are monitoring temperature already, why not add the function is heat or cool if temperature is not as needed... you could just trigger a relay that starts a fan or a the aquarium heater.
Anysways, keep up the good work! Awesome project. :) I hope wishing for functions here is no problem. If it is, I can just delete this. ;)

CelliesProjects, 2019-08-14:
Hi,
A alarm on the temperature is something I thought about.

This will for sure be after v2.0 which is the next milestone. 
Atm you have to edit config files before you can start compiling which is error prone and a pita.
v2.0 will be a single binary with all the setup functions accessible from the web interface.
Once that is done, new functionality might be added.

lullaby37, 2019-08-18:
Cool! That sounds awesome. I am really looking forward to 2.0, then. ;)
But as I said, I would not just add the alarm (which also is pretty useful if it is send via push-message for example) but I think it should also be able to trigger an event like triggering a relay for a fan or even control the fan speed...
As I said, this project is much too complex for me, so I can unfortunately not help but I really like your aquacontrol! Keep up the good work!