CelliesProjects/aquacontrol32 #34: Shematics for ILI9341

lullaby37, 2019-09-14:
Hey,

sorry to bother you again with another request. I got everything set up and working but I am still struggling with the ILI9341. I tried two different displays and several ways of wiring, yet iut does not work. 

I have bought the following displays:
https://de.aliexpress.com/item/32960934541.html
https://de.aliexpress.com/item/33012793224.html

They both work using the following example:
https://educ8s.tv/arduino-2-8-ili9341-tutorial/

Yet I cannot get them to work with your project. I think it might be duw to the wiring.
I checked wiring several time but I cannot find a mistake. I connected pins as shown in "tft_board_pins". As for the Pullups, I used a 10k resistors to connect the board pins to +3,3V. AS for TFT LED I went directly from PIN 2 using a 100Ohm resistor.
I am stuck and I cannot find my mistake, so maybe schematics would be useful. If you want, I can draw my shematics and you point out the mistakes?
Sorry to bother you again and thanks for your help.

CelliesProjects, 2019-09-17:
Schematics can be found on the web.
You could try [this gist](https://gist.github.com/CelliesProjects/b1a4252fa727856ab85e27b9976e3e48) which should test only your ILI9341.
Try to get it working without any other code/hardware and take it from there...

lullaby37, 2019-09-18:
Hey, the code from this gist gives me the following error message:

`Arduino: 1.8.9 (Windows 10), Board: "MH ET LIVE ESP32MiniKit, 80MHz, Default with spiffs, 921600, None"

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cwit\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cwit\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cwit\Documents\Arduino\libraries -fqbn=esp32:esp32:mhetesp32minikit:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\cwit\AppData\Local\Temp\arduino_build_830610 -warnings=none -build-cache C:\Users\cwit\AppData\Local\Temp\arduino_cache_431195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\cwit\Downloads\DemoPrograms\ESP32graphicstest\ESP32graphicstest.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\cwit\AppData\Local\Arduino15\packages -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\cwit\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\cwit\Documents\Arduino\libraries -fqbn=esp32:esp32:mhetesp32minikit:FlashFreq=80,PartitionScheme=default,UploadSpeed=921600,DebugLevel=none -vid-pid=10C4_EA60 -ide-version=10809 -build-path C:\Users\cwit\AppData\Local\Temp\arduino_build_830610 -warnings=none -build-cache C:\Users\cwit\AppData\Local\Temp\arduino_cache_431195 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool_py.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\esptool_py\2.6.1 -prefs=runtime.tools.mkspiffs.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\mkspiffs\0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\tools\xtensa-esp32-elf-gcc\1.22.0-80-g6c4433a-5.2.0 -verbose C:\Users\cwit\Downloads\DemoPrograms\ESP32graphicstest\ESP32graphicstest.ino
Using board 'mhetesp32minikit' from platform in folder: C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Using core 'esp32' from platform in folder: C:\Users\cwit\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.2
Detecting libraries used...
"C:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\tools\\xtensa-esp32-elf-gcc\\1.22.0-80-g6c4433a-5.2.0/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM "-DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\"" -DHAVE_CONFIG_H "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/config" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_trace" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/app_update" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/asio" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bootloader_support" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/bt" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/coap" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/console" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/driver" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp-tls" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp32" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_adc_cal" "-IC:\\Users\\cwit\\AppData\\Local\\Arduino15\\packages\\esp32\\hardware\\esp32\\1.0.2/tools/sdk/include/esp_event" "-IC:\\Users\\cwit\\Ap [... cut ...]
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ILI9341_BLACK);
  start = micros();
  for(i=min(tft.width(), tft.height()); i>20; i-=6) {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
    yield();
  }

  return micros() - start;
}`

Just to be on the save side concerning the wiring: With "10k-Pullup" you do mean a 10k transistor inbetween the esp-pin and the tft-pin that then leads to 3,3v, correct?

Thanks for your help, man!

lullaby37, 2019-09-18:
Hey man,

thanks for your help! As I said, I got it working using the example above. I will attach the code underneath.
The gist you posted cannot be compiled, there are quite a few error messages in my case, so I cannot get it working...
Yes, you are right, I could try and take it myself, but your code is just too complex for me. No idea how and what to change. I am glad that I could get the example for the tft working...

I still think it might by the wiring. Just to make sure. By saying "10k pullup" you mean a 10k resistor that connects the dataline between the esp-pin and the tft-pin to +3,3V, correct?

Thanks for your help, man!

lullaby37, 2019-09-18:
Here's the code I got my tft working...

/***************************************************
  This is our GFX example for the Adafruit ILI9341 Breakout and Shield
  ----> http://www.adafruit.com/products/1651

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

// For the Adafruit shield, these are the default.
#define TFT_DC  27
#define TFT_CLK   25
#define TFT_MOSI   32
#define TFT_MISO   39
#define TFT_CS  4
#define TFT_RST 12

// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
// If using the breakout, change pins as desired
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);

void setup() {
  Serial.begin(9600);
  Serial.println("ILI9341 Test!"); 
 
  tft.begin();

  // read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(ILI9341_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX); 
  
  Serial.println(F("Benchmark                Time (microseconds)"));
  delay(10);
  Serial.print(F("Screen fill              "));
  Serial.println(testFillScreen());
  delay(500);

  Serial.print(F("Text                     "));
  Serial.println(testText());
  delay(3000);

  Serial.print(F("Lines                    "));
  Serial.println(testLines(ILI9341_CYAN));
  delay(500);

  Serial.print(F("Horiz/Vert Lines         "));
  Serial.println(testFastLines(ILI9341_RED, ILI9341_BLUE));
  delay(500);

[... 274 lines cut ...]
  start = micros();
  for(i=0; i<w; i+=6) {
    i2 = i / 2;
    tft.drawRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(i, 0, 0));
  }

  return micros() - start;
}

unsigned long testFilledRoundRects() {
  unsigned long start;
  int           i, i2,
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ILI9341_BLACK);
  start = micros();
  for(i=min(tft.width(), tft.height()); i>20; i-=6) {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
    yield();
  }

  return micros() - start;
}

CelliesProjects, 2019-09-18:
````
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"
````
What are your library versions?

Oh and place 4 backticks ( ` ) before and after you post code, this make the code much more readable.

CelliesProjects, 2019-09-18:
````
Using library Adafruit_ILI9341 at version 1.5.1 in folder: /home/cellie/Arduino/libraries/Adafruit_ILI9341 
Using library Adafruit_GFX_Library at version 1.5.6 in folder: /home/cellie/Arduino/libraries/Adafruit_GFX_Library 
Using library ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display at version 4.0.0 in folder: /home/cellie/Arduino/libraries/ESP8266_and_ESP32_Oled_Driver_for_SSD1306_display 
Using library XPT2046_Touchscreen at version 1.3 in folder: /home/cellie/Arduino/libraries/XPT2046_Touchscreen 
Using library AsyncTCP at version 1.0.3 in folder: /home/cellie/Arduino/libraries/AsyncTCP 
Using library ESPAsyncWebServer at version 1.2.2 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/ESPAsyncWebServer 
Using library moonPhase at version 1.0.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/moonPhase 
Using library FFatSensor at version 1.0.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/FFatSensor 
Using library Task at version 1.0.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/Task 
Using library OneWire at version 2.3.3 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/OneWire 
````
These are the correct version numbers of all libs.

CelliesProjects, 2019-09-19:
This is what happens in aquacontrol:
````
...
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST );
...
...
  SPI.begin( _sclk, _miso, _mosi );
  tft.begin( 10 * 1000 * 1000 );
````
1. `Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST )` allocates pins for the breakout board TFT.
2. Then with `SPI.begin(SPI.begin( _sclk, _miso, _mosi )` the HW SPI pins are allocated.
3. After that you only have to start the tft with the bus speed as argument, as all pin values are already set.

lullaby37, 2019-09-19:
Hey,
thanks for your help and the answers.
I forgot to say that the libraries are all in the versions you have. That was actually the first thing I checked. But you are right, it is the constructor that is causing problems.

If I change the constructor in the working example code from
````
// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
// If using the breakout, change pins as desired
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
````
to
````
// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC);
// If using the breakout, change pins as desired
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
````
then the second code will only give me a white screen. So it seems that my display wants me to set all values in the constructor. Strange. Do you know any idea to make it somehow work?

Cheers! 
PS: Donated some money, please have a beer or coffee for taking the time to look into the matter.

lullaby37, 2019-09-19:
Hey,

I read that the adafruit library needs the reset pin in the constructor. I tried this, but this also only gives a white screen. The only possible way in the graphictest example is the full constructor

So this version does not work either
````
// For the Adafruit shield, these are the default.
#define TFT_DC 27
#define TFT_CLK 25
#define TFT_MOSI 32
#define TFT_MISO 39
#define TFT_CS 4
#define TFT_RST 12

// Use hardware SPI (on Uno, #13, #12, #11) and the above for CS/DC
Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_RST);
// If using the breakout, change pins as desired
//Adafruit_ILI9341 tft = Adafruit_ILI9341(TFT_CS, TFT_DC, TFT_MOSI, TFT_CLK, TFT_RST, TFT_MISO);
````

CelliesProjects, 2020-02-22:
Hi lullaby37.
Managed to solve your issue?

CelliesProjects, 2020-03-19:
Closed for now.

closed 2020-03-19 by CelliesProjects