CelliesProjects/aquacontrol32 #35: Fascinating !

ValSimDim, 2019-09-26:
Greetings and hats down for the impressive job! I recently ordered three esp32 boards and started looking for possible options to upgrade my ageing aquarium controller with esp32 based one. Soon I realized that with my very limited programming skills  I will never be able to do it myself. And today I found this brilliant work. Sorry, but I'm really excited . Donated immediately a few beers and have some questions, if you can spare some time:

1. Are you planning to add relays support in the future? I would like to control my food feeder with an option to add feed times and duration from the web interface, my self made feeder is powered by 12 volt DC motor. The real time provision is already there, so it should be an easy task. At the same time I want to be able to stop automatically all pumps in the aquarium for 30 seconds before the feeding time and start them again 2 minutes after the feeding is finished (maybe this is also a good idea to be controlled from the web interface, not hardcoded.

2.  As lullaby37 asked, it would be great if we can use the temperature readings to control another relay for heaters, or two - one for heaters above web interface set temperature and one for fans for cooling with provision for some hysteresis to avoid constant on/of cycles. 

If you can add this functionality  this would be enough for probably 90% of all freshwater aquarium hobbyists and I would definitely buy you more beer😂

I have many more ideas, but I know how hard work is all this, so ill leave them tor later, if you are willing to improve more. By the way your project is not easy to find and I'm sure if it can be somehow advertised more it will attract a lot of people. I'm starting a blog about a fully automated aquarium that can work with no attention for months, and if I build your controller I will definitely give it a separate (or more than one ) article. You can check it here: www.tufaquarium.com. I like it very much and I see a very big potential in aquacontrol32.

Cheers, Valentin

CelliesProjects, 2019-09-29:
Hi Valentin, thanks for your support!
Coding is in a bit of a slump atm because real life. 
Too busy, you know how it goes...

There is a webinterface upgrade PR running tests ( https://github.com/CelliesProjects/aquacontrol32/pull/33 ) that because of lack of time has stalled a bit.

Once that is deployed and I find some time I will start thinking/working on switching relays by temperature.

heduobacht, 2019-11-04:
Hi
I also like the Project. Looks like it is the only DIY Aquarium Controller which is state of the Art and uses cheap Hardware, but is also very functional. Still Need to get a few parts and get the logs working but i guess this will work after i looked a bit deeper into it.

Another Idea of a possible Enhancement would be the PPM/EC Meter which can be built with a few Hardwareparts which everybody should have at home. The Values can then also be put in the Graph together with the Temperature. 
It simply measures the conductivity between the Plug-wires and compensate the actual Temperature.
Only Problem on this build is, that it uses a modified OneWire Library which is not compatible to the Standard one. It is above my knowledge what they changed and why.

Here is the Project if you are interested:
https://hackaday.io/project/7008-fly-wars-a-hackers-solution-to-world-hunger/log/24646-three-dollar-ec-ppm-meter-arduino#header

Thank You for all the work you put into the Project.

Kind Regards
Daniel

closed 2020-02-09 by CelliesProjects