CelliesProjects/aquacontrol32 #38: Flashing Nodemcu

badstraw360, 2019-11-29:
Hello, I'm trying to flash a Nodemcu but I'm getting this error even if I press the flash button on the board:

´´´
A fatal error occurred: Timed out waiting for packet header
A fatal error occurred: Timed out waiting for packet header
´´´ 
What am I doing wrong? 

Thanks for putting so much effort into this project, I appreciate your help.

CelliesProjects, 2019-11-29:
I usually get this error when the flash parameters are not correct or if the partition table is overwritten.
Maybe a full flash erase (before flashing the software) might solve your problem.
````bash
~/Arduino/hardware/espressif/esp32/tools/esptool.py --port /dev/ttyUSBx erase_flash
````
Change `/dev/tty/USBx` with the port your NodeMCU is connected to.

badstraw360, 2019-11-29:
I did the full flash erase and lower flash speed but it doesn't work:

[image]

These are my settings: 

[image]

I tried to change the port where my nodeMCU is connected: 

```
~/arduino-1.8.10/arduino --board espressif:esp32:mhetesp32minikit --pref custom_DebugLevel=esp32_none --port /dev/cu.SLAB_USBtoUART  --pref upload.speed=921600 --preserve-temp-files --pref build.path=temp --pref build.partitions=default_ffat --pref build.flash_freq=80m -v --verify aquacontrol32.ino | grep  -e 'Using core' -e 'Using board' -e ' uses ' -e 'Using library' -e 'Compiling' -e 'Global'
rm gitTagVersion.h

```

and this:

```
~/arduino-1.8.10/arduino \
  --board espressif:esp32:mhetesp32minikit \
  --port /dev/cu.SLAB_USBtoUART \
  --preserve-temp-files \
  --pref build.path=temp \
  --pref upload.speed=921600 \
  --pref custom_DebugLevel=esp32_none \
  --pref build.partitions=default_ffat \
  --pref build.flash_freq=80m -v \
  --save-prefs \
  --upload aquacontrol32.ino \
```

But If I try to flash it with another code it works: 

[image]

So I think it has something to do with the settings ... 

I would really appreciate your help. :)

badstraw360, 2019-11-29:
I guess it's because the board is not ESP32... :( 

Is there a possibility to flash it with a ESP8266?

CelliesProjects, 2019-11-30:
No that will nog work.

closed 2020-02-09 by CelliesProjects