CelliesProjects/aquacontrol32 #39: Core feature, custom led program uploading does not work

Odaller, 2020-02-08:
Hello,

Probably I'm doing something wrong but I got a problem with the key feature of the aquacontrol32.
Oled works, Lights_ON works, but if i'm going in Lights_auto mode and i try to upload new timers to the controller, it doesn't do anything. If I go to another webpage and back to the editor, the timers I set before have vanished.

What could be going wrong?

Thanks for your reply.

CelliesProjects, 2020-02-08:
Most common error is a wrong partion scheme.
Did you use a partition with ffat?
If the aquacontrol file manager shows `0 Bytes free / 0 Bytes total` you probably compiled with a spiffs instead of a ffat partition.
Check `Tools>Partition Scheme` in the Arduino IDE to make sure you selected the right option.

Odaller, 2020-02-08:
Yes, that did the trick. I overlooked that in your compile notes... Thanks a lot.

commit 65517428 (2020-02-09): Show message in web interface if no ffat partition could be found.
Fixes #39
Fixes #40
closed 2020-02-08 by Odaller