CelliesProjects/aquacontrol32 #43: Moon light setup

ennuf, 2020-02-17:
**aquacontrol version: 1.6.6
**Arduino IDE version: 1..8.12
**ESP32 Arduino core version:1.0.4

The Readme tells "Moon light settings can be adjusted in the channels area of the web-interface" but I can't figure out how. Maybe the problem lies between chair and computer, I can't find any setting related to moon light on this area of the web interface that I could change. It only shows the percentage of illumonation and a picture of the moon phase.  In the lower area I can change the channel names and colors. Can you clarify how the moon light setup is supposed to work?

Thanks for your help and this great project

CelliesProjects, 2020-02-18:
It is indeed not so clear from the channels page.
The sliders on the channels page set the full moon output per channel.

The code just takes the full moon level you set and multiplies it with the percentage visible moon surface that is lit.

So:
If you put the first channel on `0.5%` then this channel will be `0.5% x 100(full moon)` = `0.5%` at full moon. 
At half moon it will be `0.5% x 0.5(half moon)` = `0.25%`. 
At new moon it will be `0.5% x 0.0(new moon)` = `0%`.

Channels set to `0%` will stay dark during the whole lunar cycle.

I hope this is clear.

CelliesProjects, 2020-02-18:
I changed the label from `bug` to `enhancement` and a future commit will address this issue.

closed 2020-03-19 by CelliesProjects

BaltasarParreira, 2022-12-06:
Hi, even reading all this I still have doubts how it works regarding the ON or OFF.
I can't see anywhere on the EDITOR to choose if is normal light or moonlight.
How really this works and I can setup specific timings for moonlight?
Thanks.