CelliesProjects/aquacontrol32 #44: Controller stuck/not properly booting after new installation

badstraw360, 2020-03-06:
**aquacontrol version:** [v1.6.6]

**Arduino IDE version:** [1.8.12]

**ESP32 Arduino core version:** [ESP32 Arduino 1.0.4 (IDF 3.2) Release]

**The issue is:**

After flashing the board, it connects to my router (espressif) through a local IP (192.168.2.25) but even if I wait 20 minutes or more for the internal flash drive to be formatted, I get a blank page. 
 I already flashed it several times after a full flash erase:
```
esptool.py --port /dev/cu.SLAB_USBtoUART  erase_flash
```
Im using the default settings to flash it and I haven't changed anything but the wifi network and password.

**Debug log:**
How can I get the debug log if I don't have access to the gui? 

**Additional information:**
Board: ESPWROOM32 - ESP32 DEVKITV1 

Btw I'm so excited because this software looks amazing and I'm so glad that you're sharing  your work. I would really appreciate your help because I'm not sure if the board has to be flashed differently. 
Thanks again.

CelliesProjects, 2020-03-06:
About the debug log:
You could set the `Core Debug Level` to something else than `none`.
See what info that gives.

CelliesProjects, 2020-03-06:
wrong button

CelliesProjects, 2020-03-06:
>#### Debug log:
>How can I get the debug log if I don't have access to the gui?

You don't need access to the gui. Debug log is visible on `Serial` aka serial port.
Use the `Serial Monitor` in the Arduino IDE. Top right icon.

But first set `Core Debug Level` in `Arduino-IDE>Tools` to `Info`, then recompile and flash with these settings and post the output here.

badstraw360, 2020-03-06:
Okay, I'm getting this: 

```
⸮[I][wifitask.ino:28] wifiTask(): Waiting 300 seconds for SmartConfig.
[I][wifitask.ino:64] wifiTask(): WiFi connected to 'HOUSE' 192.168.2.27 98:F4:AB:6D:2C:20
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
[I][webservertask.ino:688] setupMDNS(): Looking for: aquacontrol32-6D2C20.local...
[I][webservertask.ino:696] setupMDNS(): Setting up aquacontrol32-6D2C20 in mDNS.
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: timezone NOT_FOUND

```

I tried changing 

```
#define DEFAULT_HOSTNAME_PREFIX             "controller"
```

and 

```
#define COUNTRY_CODE_ISO_3166              "nl"
```

badstraw360, 2020-03-07:
I've already tried flashing the files without any modifications using another the ESP32 Dev Module configuration and I'm still getting the exact same error 

[image]

```
[I][wifitask.ino:64] wifiTask(): WiFi connected to 'HOUSE' 192.168.2.27 98:F4:AB:6D:2C:20
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
[I][webservertask.ino:688] setupMDNS(): Looking for: CONTROL-6D2C20.local...
[I][webservertask.ino:696] setupMDNS(): Setting up CONTROL-6D2C20 in mDNS.
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: timezone NOT_FOUND

```

i would really appreciate your help.

CelliesProjects, 2020-03-08:
I am trying to reproduce the issue.

badstraw360, 2020-03-08:
I've added the debug option in ``` debug.sh```  and  ``` compile.sh```

```
#!/usr/bin/env bash
if [ $(grep GIT_TAG deviceSetup.h | grep -ic TRUE) != "1" ]
then
  echo -e "\e[31mERROR\e[0m GIT_TAG not set to true in deviceSetup.h."
  exit 1
fi
currentVersion=""
if [ $(git branch | grep \* | cut -d ' ' -f2) == "master" ]
then
  currentVersion=$(git describe --tags --always)
else
  currentVersion=$(git branch | grep \* | cut -d ' ' -f2):$(git rev-parse --short HEAD)
fi
currentVersion+="!debug"
echo "const char * sketchVersion = \"$currentVersion\";" > gitTagVersion.h
echo -e "Compiling and flashing aquacontrol32 version: \e[36m$currentVersion\e[0m"
~/arduino-1.8.11/arduino \
  --board espressif:esp32:esp32 \
  --pref custom_DebugLevel=esp32_debug \
  --port /dev/ttyUSB0 \
  --pref upload.speed=921600 \
  --preserve-temp-files \
  --pref build.path=temp \
  --pref build.partitions=default_ffat \
  --pref build.flash_freq=80m -v \
  --upload aquacontrol32.ino \
  | grep  -e 'Using core' -e 'Using board' -e ' uses ' -e 'Using library' -e 'Compiling' -e 'Global'
rm gitTagVersion.h

```
```
#!/usr/bin/env bash
if [ $(grep GIT_TAG deviceSetup.h | grep -ic TRUE) != "1" ]
then
  echo -e "\e[31mERROR\e[0m GIT_TAG not set to true in deviceSetup.h."
  exit 1
fi
currentVersion=""
if [ $(git branch | grep \* | cut -d ' ' -f2) == "master" ]
then
  currentVersion=$(git describe --tags --always)
else
  currentVersion=$(git branch | grep \* | cut -d ' ' -f2):$(git rev-parse --short HEAD)
fi
echo "const char * sketchVersion = \"$currentVersion\";" > gitTagVersion.h
echo -e "Compiling aquacontrol32 version: \e[36m$currentVersion\e[0m"
~/arduino-1.8.11/arduino \
  --board espressif:esp32:esp32 \
  --pref custom_DebugLevel=esp32_debug \
  --port /dev/ttyUSB0 \
  --pref upload.speed=921600 \
  --preserve-temp-files \
  --pref build.path=temp \
  --pref build.partitions=default_ffat \
  --pref build.flash_freq=80m -v \
  --verify aquacontrol32.ino \
  | grep  -e 'Using core' -e 'Using board' -e ' uses ' -e 'Using library' -e 'Compiling' -e 'Global'
rm gitTagVersion.h

```

Then I uploaded the sketch without any modifications at all, here's the debug log:

```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[E][WiFiSTA.cpp:219] begin(): connect failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[I][wifitask.ino:28] wifiTask(): Waiting 300 seconds for SmartConfig.
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: FIND_CHANNEL
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:742] _smartConfigCallback(): Type: ESPTOUCH
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: LINK
[D][WiFiSTA.cpp:746] _smartConfigCallback(): SSID: HOUSE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.2.27, MASK: 255.255.255.0, GW: 192.168.2.1
[I][wifitask.ino:64] wifiTask(): WiFi connected to 'HOUSE' 192.168.2.27 98:F4:AB:6D:2C:20
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
[I][webservertask.ino:684] setupMDNS(): Looking for: aquacontrol32-6D2C20.local...
[I][webservertask.ino:692] setupMDNS(): Setting up aquacontrol32-6D2C20 in mDNS.
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: timezone NOT_FOUND
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: LINK_OVER
[D][WiFiSTA.cpp:755] _smartConfigCallback(): Sender IP: 192.168.2.121

```
Then uploaded the code  again in verbose mode and I get this: 
```
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[E][WiFiSTA.cpp:219] begin(): connect failed!
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[I][wifitask.ino:28] wifiTask(): Waiting 300 seconds for SmartConfig.
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 15 - AP_STACONNECTED
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: FIND_CHANNEL
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: GETTING_SSID_PSWD
[D][WiFiSTA.cpp:742] _smartConfigCallback(): Type: ESPTOUCH
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: LINK
[D][WiFiSTA.cpp:746] _smartConfigCallback(): SSID: HOUSE
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 4 - STA_CONNECTED
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 7 - STA_GOT_IP
[D][WiFiGeneric.cpp:381] _eventCallback(): STA IP: 192.168.2.27, MASK: 255.255.255.0, GW: 192.168.2.1
[I][wifitask.ino:64] wifiTask(): WiFi connected to 'HOUSE' 192.168.2.27 98:F4:AB:6D:2C:20
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: hostname NOT_FOUND
[E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
[I][webservertask.ino:684] setupMDNS(): Looking for: aquacontrol32-6D2C20.local...
[I][webservertask.ino:692] setupMDNS(): Setting up aquacontrol32-6D2C20 in mDNS.
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: timezone NOT_FOUND
[D][WiFiSTA.cpp:738] _smartConfigCallback(): Status: LINK_OVER
[D][WiFiSTA.cpp:755] _smartConfigCallback(): Sender IP: 192.168.2.121

```

It's similar to #41 , but the ```pool.ntp.org```server is working so I don't know why it's not connecting...

CelliesProjects, 2020-03-08:
Some things you might check:

1. Your ESP32 SDK, also in serial log. My dev pc has ESP32 SDK: `v3.3.1-61-g367c3c09c`
2. The version of the libraries you have installed. 

To get this info you have to enable both checkboxes in `ArduinoIDE->File->Preferences->Show verbose output during: compilation and upload.`
The output will be visible in the bottom of the Arduino IDE.

````bash
Using library SPI at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/SPI 
Using library Wire at version 1.0.1 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/Wire 
Using library FS at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/FS 
Using library FFat at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/FFat 
Using library ESPmDNS at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/ESPmDNS 
Using library Preferences at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/Preferences 
Using library WiFi at version 1.0 in folder: /home/cellie/Arduino/hardware/espressif/esp32/libraries/WiFi 
Using library Adafruit_ILI9341 at version 1.5.4 in folder: /home/cellie/Arduino/libraries/Adafruit_ILI9341 
Using library Adafruit_GFX_Library at version 1.7.5 in folder: /home/cellie/Arduino/libraries/Adafruit_GFX_Library 
Using library ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays at version 4.1.0 in folder: /home/cellie/Arduino/libraries/ESP8266_and_ESP32_OLED_driver_for_SSD1306_displays 
Using library XPT2046_Touchscreen at version 1.3 in folder: /home/cellie/Arduino/libraries/XPT2046_Touchscreen 
Using library AsyncTCP at version 1.1.1 in folder: /home/cellie/Arduino/libraries/AsyncTCP 
Using library ESPAsyncWebServer at version 1.2.3 in folder: /home/cellie/Arduino/libraries/ESPAsyncWebServer 
Using library OneWire at version 2.3.3 in folder: /home/cellie/Arduino/libraries/OneWire 
Using library moonPhase at version 1.0.0 in folder: /home/cellie/Arduino/libraries/moonPhase 
Using library FFatSensor at version 1.0.2 in folder: /home/cellie/Arduino/libraries/FFatSensor 
Using library Task at version 1.0.0 in folder: /home/cellie/Arduino/libraries/Task 
````
My guess is DNS resolution is somehow wonky at your end.
I don't know how to check for DNS resolution on the ESP32 and will have to dive into that.
Stay tuned.

badstraw360, 2020-03-08:
Libraries used: 
```
Using library SPI at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/SPI 
Using library Wire at version 1.0.1 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Wire 
Using library FS at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FS 
Using library FFat at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/FFat 
Using library ESPmDNS at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/ESPmDNS 
Using library Preferences at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/Preferences 
Using library WiFi at version 1.0 in folder: /Users/memo/Library/Arduino15/packages/esp32/hardware/esp32/1.0.4/libraries/WiFi 
Using library Adafruit_ILI9341 at version 1.5.4 in folder: /Users/memo/Documents/Arduino/libraries/Adafruit_ILI9341 
Using library Adafruit_GFX_Library at version 1.7.5 in folder: /Users/memo/Documents/Arduino/libraries/Adafruit_GFX_Library 
Using library esp8266-oled-ssd1306 at version 4.1.0 in folder: /Users/memo/Documents/Arduino/libraries/esp8266-oled-ssd1306 
Using library XPT2046_Touchscreen at version 1.3 in folder: /Users/memo/Documents/Arduino/libraries/XPT2046_Touchscreen 
Using library AsyncTCP at version 1.1.1 in folder: /Users/memo/Documents/Arduino/libraries/AsyncTCP 
Using library ESPAsyncWebServer at version 1.2.3 in folder: /Users/memo/Documents/Arduino/libraries/ESPAsyncWebServer 
Using library OneWire at version 2.3.3 in folder: /Users/memo/Documents/Arduino/libraries/OneWire 
Using library moonPhase at version 1.0.0 in folder: /Users/memo/Documents/Arduino/libraries/moonPhase 
Using library FFatSensor at version 1.0.2 in folder: /Users/memo/Documents/Arduino/libraries/FFatSensor 
Using library Task at version 1.0.0 in folder: /Users/memo/Documents/Arduino/libraries/Task
```

Full verbose output during compilation and upload:

```
/private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/hardware -hardware /Users/memo/Library/Arduino15/packages -tools /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/memo/Library/Arduino15/packages -built-in-libraries /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/libraries -libraries /Users/memo/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=defaultffat,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose -vid-pid=0000_0000 -ide-version=10812 -build-path /var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/arduino_build_667084 -warnings=none -build-cache /var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/arduino_cache_472875 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.mkspiffs.path=/Users/memo/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.mkspiffs-0.2.3.path=/Users/memo/Library/Arduino15/packages/esp32/tools/mkspiffs/0.2.3 -prefs=runtime.tools.xtensa-esp32-elf-gcc.path=/Users/memo/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.xtensa-esp32-elf-gcc-1.22.0-80-g6c4433a-5.2.0.path=/Users/memo/Library/Arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/1.22.0-80-g6c4433a-5.2.0 -prefs=runtime.tools.esptool_py.path=/Users/memo/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -prefs=runtime.tools.esptool_py-2.6.1.path=/Users/memo/Library/Arduino15/packages/esp32/tools/esptool_py/2.6.1 -verbose /Users/memo/Downloads/aquacontrol32/aquacontrol32.ino
/private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/hardware -hardware /Users/memo/Library/Arduino15/packages -tools /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/tools-builder -tools /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/hardware/tools/avr -tools /Users/memo/Library/Arduino15/packages -built-in-libraries /private/var/folders/0w/z238yjs90zz9g4rfyxsnlvnm0000gn/T/AppTranslocation/5A134B2C-B0B6-4E22-9F92-88EB718BD0A4/d/Arduino.app/Contents/Java/libraries -libraries /Users/memo/Documents/Arduino/libraries -fqbn=esp32:esp32:esp32:PSRAM=disabled,PartitionScheme=defaultffat,CPUFreq=240,FlashMode=qio,FlashFreq=80,FlashSize=4M,UploadSpeed=921600,DebugLevel=verbose -vid-pid=0000_0000 -ide-version=10812 -build-path /var/folders/0w/z238yjs90zz9 [... cut ...]
Writing at 0x000a0000... (97 %)
Writing at 0x000a4000... (100 %)
Wrote 1124608 bytes (615653 compressed) at 0x00010000 in 9.3 seconds (effective 968.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 127...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (127 compressed) at 0x00008000 in 0.0 seconds (effective 1933.5 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

```

ESP32 SDK version:

[image]

Apparently I have a lower SDK version than yours.. Would that be an issue?

badstraw360, 2020-03-09:
Im getting this after I compile and install the latest arduino on a windows computer:

 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
[I][aquacontrol32.ino:270] setup(): aquacontrol32 v1.6.6-master
[I][aquacontrol32.ino:271] setup(): ESP32 SDK: v3.2.3-14-gd3e562907
[I][aquacontrol32.ino:306] setup(): No ILI9341 found
[I][aquacontrol32.ino:311] setup(): No SSD1306 OLED found.
[I][aquacontrol32.ino:352] setup(): FFat mounted.
[I][wifitask.ino:64] wifiTask(): WiFi connected to 'CASA' 192.168.2.27 98:F4:AB:6D:2C:20
[E][ESPmDNS.cpp:72] begin(): Failed setting MDNS hostname
[I][webservertask.ino:688] setupMDNS(): Looking for: aquacontrol32-6D2C20.local...
[I][webservertask.ino:696] setupMDNS(): Setting up aquacontrol32-6D2C20 in mDNS.
[E][Preferences.cpp:437] getString(): nvs_get_str len fail: timezone NOT_FOUND

CelliesProjects, 2020-03-09:
Does this work on your esp32?
 It is a simple sketch that just connects to a NTP server and syncs the clock.

````c++
// NTP test sketch
#include <WiFi.h>

const char * WIFI_NETWORK  = "";
const char * WIFI_PASSWORD = "";

const char * TIME_ZONE   = "CET-1CEST-2,M3.5.0/2,M10.5.0/3";  //https://remotemonitoringsystems.ca/time-zone-abbreviations.php
const char * TIME_SERVER = "nl.pool.ntp.org";

void setup() {
  Serial.begin(115200);
  if (!strlen(WIFI_NETWORK)) {
    Serial.println("No WiFi defined");
    while (1) delay(100);
  }
  Serial.println("Connecting...");
  WiFi.begin(WIFI_NETWORK, WIFI_PASSWORD);

  while (!WiFi.isConnected())
    delay(10);

  Serial.println("Syncing NTP...");
  configTzTime(TIME_ZONE, TIME_SERVER);
  struct tm timeinfo = {0};
  while (!getLocalTime(&timeinfo, 0))
    vTaskDelay(10 / portTICK_PERIOD_MS);

  Serial.println("We have NTP sync.");
}

void loop() {
  // put your main code here, to run repeatedly:
  static time_t savedTime;
  time_t now = time(NULL);
  if (now != savedTime) {
    struct tm localTime;
    localtime_r(&now, &localTime);
    char timestr[40];
    strftime(timestr, sizeof(timestr), "%c", &localTime);
    Serial.printf("Current time: %s\n", timestr);
    savedTime = now;
  }
  delay(10);
}
````

badstraw360, 2020-03-09:
I get this: 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Connecting...
Syncing NTP...

```
 
So apparently the NTP server isn't syncing...

badstraw360, 2020-03-09:
If I try this code: 

```c
#include <WiFi.h>
#include "time.h"

const char* ssid       = "SSID";
const char* password   = "PASSWORD";

const char* ntpServer = "pool.ntp.org";
const long  gmtOffset_sec = 3600;
const int   daylightOffset_sec = 3600;

void printLocalTime()
{
  struct tm timeinfo;
  if(!getLocalTime(&timeinfo)){
    Serial.println("Failed to obtain time");
    return;
  }
  Serial.println(&timeinfo, "%A, %B %d %Y %H:%M:%S");
}

void setup()
{
  Serial.begin(115200);
  
  //connect to WiFi
  Serial.printf("Connecting to %s ", ssid);
  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
      delay(500);
      Serial.print(".");
  }
  Serial.println(" CONNECTED");
  
  //init and get the time
  configTime(gmtOffset_sec, daylightOffset_sec, ntpServer);
  printLocalTime();

  //disconnect WiFi as it's no longer needed
  WiFi.disconnect(true);
  WiFi.mode(WIFI_OFF);
}

void loop()
{
  delay(1000);
  printLocalTime();
}
```

I get this: 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Connecting to HOUSE .. CONNECTED
Failed to obtain time
Failed to obtain time
Failed to obtain time
Failed to obtain time
Failed to obtain time

```

So yeah , it's definitely not syncing..

CelliesProjects, 2020-03-09:
Could you test it on another network?

badstraw360, 2020-03-10:
Not right now, but I'll see if I can go to a starbucks or something like that. Would it be hard to find an alternative to set the date and time?

CelliesProjects, 2020-03-11:
Just copy-pasted a static IP setup sketch.
Setting a static IP gives you the option to set the dns servers. I think the problem is that the dns server ip addresses are not passed on by your dhcp server.

It is on the to-do list for aquacontrol to make an an option for a static IP which makes more sense for a webserver than dhcp anyway.

Could you try this? See if this kind of setup would solve your problem.

Just change the desired local_IP, gateway IP, subnet if necessary and the dns servers. Your upstream dns servers can be found somewhere in your router or you could use just for testing [Google dns](https://developers.google.com/speed/public-dns/). 8.8.8.8 and 8.8.4.4

````c++
// Load Wi-Fi library
#include <WiFi.h>

// Replace with your network credentials
const char* ssid     = "";
const char* password = "";

const char * TIME_ZONE   = "CET-1CEST-2,M3.5.0/2,M10.5.0/3";  //https://remotemonitoringsystems.ca/time-zone-abbreviations.php
const char * TIME_SERVER = "nl.pool.ntp.org";

void setup() {
  Serial.begin(115200);

  IPAddress local_IP(192, 168, 0, 60);  /* THIS SHOULD BE OUTSIDE YOUR ROUTERS DHCP RANGE! */

  IPAddress gateway(192, 168, 0, 1);   // Set your Gateway IP address
  IPAddress subnet(255, 255, 255, 0);   /* USUALLY 255,255,255,0 BUT CHECK IT IN YOUR ROUTER OR PC CONNECTED TO THE SAME ROUTER*/
  IPAddress primaryDNS(192, 168, 0, 30); // CHECK IN YOUR ROUTER
  IPAddress secondaryDNS( 192, 168, 0, 50 ); // CHECK IN YOUR ROUTER

  Serial.printf("Connecting to: %s\n", ssid);

  WiFi.begin(ssid, password);

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println();

  // Configures static IP address
  if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.print( "STA Failed to configure" );
    while (1) delay(10);
  }

  // Print local IP address
  Serial.printf("Local IP: %s\n", WiFi.localIP().toString().c_str());

  // Print ESP MAC Address
  Serial.printf( "MAC address: %s\n", WiFi.macAddress().c_str());

  Serial.println("Syncing NTP...");
  configTzTime(TIME_ZONE, TIME_SERVER);
  struct tm timeinfo = {0};
  while (!getLocalTime(&timeinfo, 0))
    vTaskDelay(10 / portTICK_PERIOD_MS);

  Serial.println("We have NTP sync.");
}

void loop() {
  static time_t savedTime;
  time_t now = time(NULL);
  if (now != savedTime) {
    struct tm localTime;
    localtime_r(&now, &localTime);
    char timestr[40];
    strftime(timestr, sizeof(timestr), "%c", &localTime);
    Serial.print(&localTime,"Current time: %c\n");
    savedTime = now;
  }
  delay(10);
}
````

badstraw360, 2020-03-12:
Amazing it works!!!

```

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:8896
load:0x40080400,len:5816
entry 0x400806ac
Connecting to: HOUSE
.
Local IP: 192.168.2.27
MAC address: 98:F4:AB:6D:2C:20
Syncing NTP...
We have NTP sync.
Current time: Thu Mar 12 20:27:54 2020
Current time: Thu Mar 12 20:27:55 2020
```

CelliesProjects, 2020-03-13:
Hey that's great.

If you want to integrate it yourself, cool!
How? That's the art of programming.

This is just a demo/test sketch to see what the problem really is.
To integrate it in aquacontrol I will have to solve all possible use cases like how to reconnect etc..
I am taking some time to think over this issue and maybe even produce some code this weekend.
I do accept PR's if you come up with some solution I like.

CelliesProjects, 2020-03-13:
You could pull #45 and see if that works for you.

Set the options in `aquacontrol32.ino` and flash as usual.

badstraw360, 2020-03-14:
Omg this is amazing !!! It works !! Thank you so much !

I’ve been trying to save the channel settings in safari but it gives me an “not authorized” error message, unless I use chrome.
I’m gonna test it a little more and report back..

Thanks again for your patience and sharing your code.

CelliesProjects, 2020-03-15:
Good to hear. This week I will test #45 and merge it in master if no problems come up.
There will be a new release after a successfull merge so stay tuned.
Closing this issue for now.

closed 2020-03-15 by CelliesProjects