CelliesProjects/aquacontrol32 #49: Description off /webif/Readme.md Wrong

crazybanane, 2021-12-18:
Hello,

i tried to make a Language change for the Html files and convert it to a same lookin file as your e.g. channels_htm_gz.h
But when i will compile it Arduino gives me failure on compiling. I think because i didnt gzipped it. But how to gzip it that it will work?
Or in what order i have to zip the file?

closed 2021-12-19 by CelliesProjects

CelliesProjects, 2021-12-19:
Hi @crazybanane 

Thanks for your interest in aquacontrol and pointing this out.

The explanation was outdated.
I updated the description on preparing the web interface files. Hopefully this made it more clear.

[Updated explanation](/tree/master/webif)

Feel free to reopen this issue if things dont work out or are unclear.

crazybanane, 2021-12-19:
Thank you so much. It works just fine also on WIN11 with WSL and Kali Linux.
Is there a way that i can Upload German Language Website file inclusive Gzipped files here that anybody can use it?
Thank you so much for this great and Easy made Program

CelliesProjects, 2021-12-19:
Yes, a PR with German language translation would be a nice addition.

I do have to think a bit about how to make it work in an easy and maintainable way.
I will report back if I have worked something out.

crazybanane, 2021-12-19:
I took the files from webif and translated it to german i can send you these if you want.

CelliesProjects, 2021-12-19:
Yes, you can send them to aquacontrol @ wasietsmet.nl.

crazybanane, 2021-12-19:
> Yes, you can send them to aquacontrol @ wasietsmet.nl.

On the way

crazybanane, 2021-12-19:
Is it easy enough if you made in webif new folder with the language e.g.
webif>
         -de
         -en
         -esp........

and correct the code that the Compiler can search for it when language is set?
Or is it possible to upload the wbesite files via filemanager?

CelliesProjects, 2021-12-20:
@crazybanane 
I might go for something like the first approach.
The second approach exposes the files through the filemanager and is imho not the best (rather ugly) solution.
For now I got multiple other fish to fry.

Thanks for your time and effort!

commit 202cef8c (2021-12-19): Update README.md. Should fix #49.