CelliesProjects/aquacontrol32 #51: Pindefinition not Clear

crazybanane, 2021-12-28:
Hey, 
i am on the way to make for your Project an Fritzing Connection scheme but actually i have a Problem with PIN 12 in Master/devicePinSetup i see the following line

#define SPI_TFT_RST_PIN                     12  /* Goes to TFT RESET */

but i cant find anywhere a description of that pin... not on the Pinout on the web or in the PIN-Description of the Arduino ESP 1.0.4 Core Package. can you please describe or change the title to the Original Board Marking like so:

#define SPI_SCK_PIN                         25  /* Goes to TFT SCK/CLK, Board PIN  IO25*/
#define SPI_MOSI_PIN                        32  /* Goes to TFT MOSI, Board PIN IO32 */
#define SPI_TFT_RST_PIN                     12  /* Goes to TFT RESET, Board PIN ??GPIO12/TDI?? */
#define SPI_TFT_CS_PIN                      4   /* Goes to TFT CS,Board PIN IO4 */
#define SPI_SD_CS_PIN                       0   /* Goes to SD CS,Board PIN IO0 */
#define SPI_MISO_PIN                        39  /* Goes to TFT MISO,Board PIN ??SVN/GPIO39?? */
#define TOUCH_CS_PIN                        33  /* Goes to TFT T_CS,Board PIN IO33 */
#define TOUCH_IRQ_PIN                       35  /* Goes to TFT T_IRQ, Board PIN IO35 */

That willl be very helpfull, thank for your Hard work.
If iam ready then you can upload something like this:
[image]

Actually this is only Prototyping not the endproduct.

CelliesProjects, 2021-12-29:
Hi @crazybanane.

You are right, on these boards there is no `GPIO12` marking.
On my board it is marked `TDI`. 

[image]

I put `GPIO12` there as the pinout (and silkscreen) might change from board to board.

Hope this helps.

CelliesProjects, 2021-12-29:
@crazybanane 
One thing I saw when testing my board with a scope was a voltage surge on the mosfet drains every time the mosfet stopped conducting. I dont recall the exact numbers, but it was quite a high voltage and it is a function of the amount of current that is suddenly stopped. 
I solved this by putting a n4001 diode between the mosfet drain and +12V with the cathode on the +12V side. 
This will limit this surge to about 0.7V.
And use a beefy elco (or elcos) to absorb this current.

The diodes are on the bottom in the image. 

[image]

crazybanane, 2021-12-30:
Hey thanks for answer,

i cant take a look for ringing by my Board because i dont have an Oscillioscope. Actual i dont have Problems with Dimming or Failures like Brownouts. At the moment i have only 1 COB Panel with 70 Watt on Channel 1 but it works super Smooth and Great.
I take between ESP Pin for Channel 1 an 220Ohm resistor to Gate and an 10k to GND.  On an older Board i had problems with 470Ohms but it was another software. I will test later some other Leds that i have. I think in the near Future i will buy an simple Osci but i dont have too much of plan to work so detailed.

So i think in finished the board for other DIY-people so that they can easier understand the connections if you find failures or big Problems please write me otherwise you can take this to your Project. Hopefully it can Help.

I will correct these Days the German Language files because i found some text that is not suitable, you will got then mail from me.

[image]

So then i wish you a Happy New Year or Gelukkig nieuwjaar

CelliesProjects, 2021-12-31:
@crazybanane guten Rutsch!