CelliesProjects/aquacontrol32 #54: How Moonlight ON or OFF works?

BaltasarParreira, 2022-12-07:
Hi, even reading all this (https://github.com/CelliesProjects/aquacontrol32/issues/43) I still have doubts how it works regarding the ON or OFF.
I can't see anywhere on the EDITOR to choose if is normal light or moonlight.
How really this works and I can setup specific timings for moonlight?
Thanks.

CelliesProjects, 2022-12-08:
Hi @BaltasarParreira. The sliders on the channel page set the percentage at full moon. 
So if you set it to 0.99% for a channel then at full moon that channel will be at 0.99%. 
For the rest of the month the value set is just proportional to that.
So at new moon it will be at 0% and if the moon is half full the percentage will be half of 99%. And when the moon is lit for 25% then the lights will be set to 0.99 x 0.25 which is  0.24%.

How low you can go with the light settings depends very much on the used led strip. Some that I used or tested stop at 0.5% -they just switch off- while others will go a lot lower. Atm I have the moonlicht setting at 0.06% and still a reasonable gradation over the month, but not all led strips will go that low.

BaltasarParreira, 2022-12-08:
Hi, thanks. but I know all that and understand how it works regarding the intensity and the calculation based on a full mon day.
My doubts are about the ON and OFF like when or what I have to do on my timers so the moonlight kicks on after the normal percentage setup for a timer/day.
Right now I have this setup for my blue channel so I ask on this setup when the moonlight kicks on and until what time:
[image]

CelliesProjects, 2022-12-09:
There is no kicking in. The moonlight is implemented as do-not-dim-below-current-moonlight.

BaltasarParreira, 2022-12-09:
Sorry I didn't understand what you mean !?!?
Can you provide a time scheduler example and indicate on that when I can see moonlight on or in what time the moon percentage will be the one for my setting from my Blue channel for example (0,30% that I setup for full moon)?

CelliesProjects, 2022-12-21:
I'm sorry. No idea. But most leds need at least something like 1-2V to function. About other lighting systems I don't know.

LZsolt75, 2024-05-22:
How can I increase the lighting level at the sliders on the channels page? My leds start to lit at 2.8% so something like 3% at full moon would be good for me.

CelliesProjects, 2024-05-23:
@LZsolt75 I think you could hack the sliders on the channels web page.

See https://github.com/CelliesProjects/aquacontrol32/blob/master/webif/channels.htm#L126-L132

`<div class="item"><input type="text" class="channelName"><input type="color" class="colorPicker"><input type="range" class="minLevel" value="0" min="0" max="0.99" step="0.01"><p class="itemValue">0%</p></div>`

would become 

`<div class="item"><input type="text" class="channelName"><input type="color" class="colorPicker"><input type="range" class="minLevel" value="0" min="0" max="4.00" step="0.01"><p class="itemValue">0%</p></div>`

for all 5 pickers. -Or the ones you want to use for moonlight-
While you are at it maybe also increase the step size.

You will then need to convert the edited htm file twice, once to gzip the content and then convert that to a `.h` file.
The procedure is explained in detail on https://github.com/CelliesProjects/aquacontrol32/tree/master/webif