CelliesProjects/aquacontrol32 #55: Screen is only white

LadaSi, 2023-01-26:
Hello.The controller is working perfectly but the ILI9341 screen is only white and I need some advice.Thanks

CelliesProjects, 2023-01-27:
Hmm the ili9341 is a real pita to get going. The build quality is very poor and there are a lot of differences between the different boards from suppliers like aliexpress, ebay etc.
I am working on a v2 and I will not add a ili9341 this time to save myself a lot of hassle.

Does the display work with the examples from the ili9341 library?
Did you try the force option in the web interface?
Are there pullups on the relevant pins from the display? Without these the display will not work for very long if at all.
Also check the power supply, feeding it from the 3v3 on most esps will not work properly. Use a seperate LDO for the display.

CelliesProjects, 2023-01-27:
See also https://forum.arduino.cc/t/ili9341-white-screen-sleep-mode-caused-by-unstable-power/322737

LadaSi, 2023-01-28:
Hello. So I found and tried the attached test code on the MH-ET LIVE MiniKit ESP32 and my display works with this code and pinout. So please advise what to do next. Thank you very much for now

/***************************************************
  This is our GFX example for the Adafruit ILI9341 Breakout and Shield
  ----> http://www.adafruit.com/products/1651

  Check out the links above for our tutorials and wiring diagrams
  These displays use SPI to communicate, 4 or 5 pins are required to
  interface (RST is optional)
  Adafruit invests time and resources providing this open source code,
  please support Adafruit and open-source hardware by purchasing
  products from Adafruit!

  Written by Limor Fried/Ladyada for Adafruit Industries.
  MIT license, all text above must be included in any redistribution
 ****************************************************/

#include "SPI.h"
#include "Adafruit_GFX.h"
#include "Adafruit_ILI9341.h"

#define _cs   4  // goes to TFT CS
#define _dc   27  // goes to TFT DC
#define _mosi 32  // goes to TFT MOSI
#define _sclk 25  // goes to TFT SCK/CLK
#define _rst  12   // goes to TFT RESET
#define _miso 39    // Not connected

#define TFT_BACKLIGHT_PIN  2
//       3.3V     // Goes to TFT LED  
//       5v       // Goes to TFT Vcc
//       Gnd      // Goes to TFT Gnd        

// Use hardware SPI
//Adafruit_ILI9341 tft = Adafruit_ILI9341(_cs, _dc, _rst);

// If using software SPI change pins as desired
Adafruit_ILI9341 tft = Adafruit_ILI9341(_cs, _dc, _mosi, _sclk, _rst);

void setup() {
  Serial.begin(115200);
  Serial.println("ILI9341 Test!"); 

  pinMode( TFT_BACKLIGHT_PIN, OUTPUT );
  digitalWrite( TFT_BACKLIGHT_PIN, HIGH );
 
  tft.begin();tft.setRotation(3);tft.fillScreen(ILI9341_BLACK); tft.setTextSize(3);tft.println("Starting...");delay(1000);
  tft.fillScreen(ILI9341_BLACK);
  // read diagnostics (optional but can help debug problems)
  uint8_t x = tft.readcommand8(ILI9341_RDMODE);
  Serial.print("Display Power Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDMADCTL);
  Serial.print("MADCTL Mode: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDPIXFMT);
  Serial.print("Pixel Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDIMGFMT);
  Serial.print("Image Format: 0x"); Serial.println(x, HEX);
  x = tft.readcommand8(ILI9341_RDSELFDIAG);
  Serial.print("Self Diagnostic: 0x"); Serial.println(x, HEX); 
  
  Serial.println(F("Benchmark                Time (microseconds)"));
  delay(10);
  Serial.print(F("Screen fill              "));
  Serial.println(testFillScreen());
  delay(500);

  Serial.print(F("Text                     "));
  Serial.println(testText());
  delay(500);

[... 282 lines cut ...]
  start = micros();
  for(i=0; i<w; i+=6) {
    i2 = i / 2;
    tft.drawRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(i, 0, 0));
  }

  return micros() - start;
}

unsigned long testFilledRoundRects() {
  unsigned long start;
  int           i, i2,
                cx = tft.width()  / 2 - 1,
                cy = tft.height() / 2 - 1;

  tft.fillScreen(ILI9341_BLACK);
  start = micros();
  for(i=min(tft.width(), tft.height()); i>20; i-=6) {
    i2 = i / 2;
    tft.fillRoundRect(cx-i2, cy-i2, i, i, i/8, tft.color565(0, i, 0));
    yield();
  }

  return micros() - start;
}

closed 2024-05-16 by LadaSi