CelliesProjects/aquacontrol32 #58: Arduino IDE verify problem

tibike32, 2023-10-03:
**aquacontrol version:** 1.6.8

**Arduino IDE version:** 2.2.1

**ESP32 Arduino core version:** ESP32 DEV module

**The issue is:**
[replace this with a clear and concise desciption of the issue]

**Debug log:**
C:\aquacontrol32-1.6.8\aquacontrol32\webservertask.ino: In function 'bool setupMDNS(const char*)':
C:\aquacontrol32-1.6.8\aquacontrol32\webservertask.ino:673:49: error: cannot convert 'ip4_addr*' to 'esp_ip4_addr_t*' {aka 'esp_ip4_addr*'}
   esp_err_t res = mdns_query_a( hostname, 2000, &addr );
                                                 ^~~~~
In file included from C:\Users\tberenyi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11\libraries\ESPmDNS\src/ESPmDNS.h:46,
                 from C:\aquacontrol32-1.6.8\aquacontrol32\aquacontrol32.ino:7:
C:\Users\tberenyi\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.11/tools/sdk/esp32/include/mdns/include/mdns.h:649:83: note:   initializing argument 3 of 'esp_err_t mdns_query_a(const char*, uint32_t, esp_ip4_addr_t*)'
 esp_err_t mdns_query_a(const char * host_name, uint32_t timeout, esp_ip4_addr_t * addr);
                                                                  ~~~~~~~~~~~~~~~~~^~~~

exit status 1

Compilation error: cannot convert 'ip4_addr*' to 'esp_ip4_addr_t*' {aka 'esp_ip4_addr*'}

**Additional information:**
I tried with another Arduino IDE version, but the problem is the same

CelliesProjects, 2023-10-03:
@tibike32 I think this is because the ESP32 Arduino Core version you use is too recent.
The mdns lib changed quite a lot after I think 1.0.5 or 1.0.6.

Try compiling with [ESP Arduino Core 1.0.4](https://github.com/espressif/arduino-esp32/releases/tag/1.0.4) and see if that helps. 
Also take a look at which [libraries](https://github.com/CelliesProjects/aquacontrol32#libraries) need which version.
Most have changed considerably since 2019.

tibike32, 2023-10-03:
If I use the Arduino IDE 1.8.12, still this error message. But I can't how use it the ESP Arduino Core 1.0.4 ?
I'm basic Arduino user.

tibike32, 2023-10-03:
> @tibike32 I think this is because the ESP32 Arduino Core version you use is too recent. The mdns lib changed quite a lot after I think 1.0.5 or 1.0.6.
>
> Try compiling with [ESP Arduino Core [...]

If I use the Arduino IDE 1.8.12, still this error message. But I can't how use it the ESP Arduino Core 1.0.4 ?
I'm basic Arduino user.
I check the libraries thats all ok.

CelliesProjects, 2023-10-03:
Go to boards manager and search for esp32. 

Then click 1.0.4.

[image]

tibike32, 2023-10-03:
> Go to boards manager and search for esp32.
> 
> Then click 1.0.4.
> 
> I found it, but I just can't find it here 2.0.10 and 2.0.11

tibike32, 2023-10-03:
> Go to boards manager and search for esp32.
> 
> Then click 1.0.4.
> 
> 

That's great !!! The verify is ok . 
Thank you the help !
Good night !

CelliesProjects, 2023-10-03:
Closing this issue. Feel free to re-open if needed.

closed 2023-10-03 by CelliesProjects